martinbogo/pt-battery-diagnostics #1: Its mid January

rhinomods, 2019-01-21:
I know there are a bunch of people who are as excited as myself to see  and contribute to this github, do you have any thing to upload?

martinbogo, 2019-01-22:
I'm only able to devote weekend hours to the project at the moment.  Unfortunately, the Segway rental dealer here in Austin TX closed down, and I lost access to the pile of used batteries I was using to test my code and hardware.   

I went ahead and bought a couple good used batteries in December, which I'm using to develop the hardware and code, along with a Segway i2 1st generation, so things should speed up now.

felixfetche, 2019-01-22:
Okay,
Wish you time and concentration to go further!

Trimis de pe iPad‑ul meu

felixfetche, 2019-01-23:
5000 €

Trimis de pe iPad‑ul meu

martinbogo, 2019-01-23:
Felix,

If you will be participating in this project it all has to be open-source
and free of IP encumberance.  Otherwise, I will have to deny you access to
checking in code and information to prevent IP tainting.

I have a set of boards coming from PCB Express.  Once I verify that
everything is working as expected, I will be uploading the gerbers, BOM,
arduino code and STL files for the plug adapter.

As of now I can read all cell voltages and decode battery serial number and
state messages.

Sincerely,
-Martin

felixfetche, 2019-01-23:
With all respect,
Apreciate your intentions, but you have to show some results about your research job. To see that you can contribute. Otherwise cannot talk level to level .Agree?

Trimis de pe iPad‑ul meu

krex4, 2019-01-24:
Ok, i do maintenance on Segway Batteries since 2008, and i do it mannualy, open case ( with care) , then test cells , change leaked ( bad) cells, then recharge pack,
I test it on a 2A load , then for 5 min or 10 on a 4A load , always testing cells.
I mod a onboard charger ( gen 1) to charge on my bench.
All tests done with my Fluke, i’ve seen on forum a software that a Japanese member has that reset’s the board on battery, and find it interesting, then i saw on forum that you ( sir) will post a schematic and arduino sketck of a i2c interface ( i presume wit 2  2 K ohms resistors to +5v ) to measure cells , and maybe reset the board.
It is really a good tool for my bench, then i will wait for your research ,

Thank you,

Regards.

Enviado do meu iPhone

rhinomods, 2019-01-24:
can you elaborate more on the on-board charger modification. ( the CSB of a Gen1 machine) 
From my poking around the off-board charger does not use any of the i2c communication (j6 )(j7)
 nor the + power line going into the battery.
The off board charger Only through pins 3(j4) 9(J3), 11(j1) ,12(j5)
I am assuming your On-Board charger is similar.

martinbogo, 2019-01-24:
> Yes , i use a CSB of a Gen 1 Segway to offboard charge my batteries, i use
> the 4 pins described , but on one pin i use a 100 ohm resistor and a diode
> (1N 4007) in series.
>
> Enviado do meu iPhone

Speaking of pins, the diagram I uploaded should probably be updated with
the pin names, and not just the functions.

For this project, I'm concentrating on the lithium-ion batteries since they
use the i2c communication.   I don't have, and have not been able to play
with, the older NiMH cells.

felixfetche, 2019-01-24:
Let’s talk when you get close to the subject, sofar you are so far...

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: Martin Bogomolni <notifications@github.com>
Trimis: joi, ianuarie 24, 2019 10:03 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] Its mid January (#1)

Speaking of pins, the diagram I uploaded should probably be updated with
the pin names, and not just the functions.

For this project, I'm concentrating on the lithium-ion batteries since they
use the i2c communication. I don't have, and have not been able to play
with, the older NiMH cells.

martinbogo, 2019-01-24:
Felix,

You're not adding anything constructive to this discussion.  I'm locking/closing this topic.  This is GitHub after all and not a general discussion forum.    If you want to generally discuss buying/selling your research, I suggest taking this to SegwayChat, YouTube, Twitter, or a different forum.

Sincerely,
Martin

martinbogo, 2019-01-27:
Code has been uploaded to read the voltages in the battery.  I'm closing this issue and marking it as resolved.

martinbogo, 2019-01-27:
Issue has been resolved.

closed 2019-01-27 by martinbogo