martinbogo/pt-battery-diagnostics #3: LCD on Arduino To read values

krex4, 2019-02-19:
Is it possible to read vealues of measurements  on  a LCD ?

felixfetche, 2019-02-19:
https://youtu.be/YHkUdpllZM8

Trimis de pe iPhone‑ul meu

krex4, 2019-02-19:
Great, 500 Euros is cheap , maybe i will buy 10 😁

Enviado do meu iPhone

https://youtu.be/YHkUdpllZM8

felixfetche, 2019-02-19:
Learn to be profesional, don’t act childlish.
Keep your jokes with your coleagues at school.

Is not 500€, is more cheap, the DIY kit.

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: krex4 <notifications@github.com>
Trimis: marți, februarie 19, 2019 6:59 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Great, 500 Euros is cheap , maybe i will buy 10 😁

Enviado do meu iPhone

https://youtu.be/YHkUdpllZM8

martinbogo, 2019-02-19:
Fairly trivial ... sure.  I'll can pretty easily add support to use an i2c
display, something inexpensive.

These are $5-10US and available world-wide.

https://www.amazon.com/gp/product/B07D7ZQVDR/

martinbogo, 2019-02-19:
Speaking of being professional,

Felix, please stop advertising your product, unless you are going to
voluntarily add code to this project.

Sincerely,
Martin

felixfetche, 2019-02-19:
I’m not advertising, i just respond to you, otherwise contact me privately please.

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: Martin Bogomolni <notifications@github.com>
Trimis: marți, februarie 19, 2019 9:15 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Speaking of being professional,

Felix, please stop advertising your product, unless you are going to
voluntarily add code to this project.

Sincerely,
Martin

krex4, 2019-02-19:
Ah ok 😁
Sorry about my joke, please
forgive-me!
I have the Arduino Uno, LCD ( oled ) , and board connector.
I need the code and connections schematics.
How much you ask for them ?

Regards,

Enviado do meu iPhone

De la: krex4 <notifications@github.com<mailto:notifications@github.com>>

https://youtu.be/YHkUdpllZM8

krex4, 2019-02-19:
Ok , thank you 👍😁

Enviado do meu iPhone

https://www.amazon.com/gp/product/B07D7ZQVDR/

felixfetche, 2019-02-19:
Is not important the display thing, ...
Retriving correctly the datas from BCM and then filter and process them is the key of the job...

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: krex4 <notifications@github.com>
Trimis: marți, februarie 19, 2019 11:23 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Ok , thank you 👍😁

Enviado do meu iPhone

https://www.amazon.com/gp/product/B07D7ZQVDR/

felixfetche, 2019-02-19:
First byte, the CRC is almost most important, not easy, that’s why i do not let it for free

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: felix Fetche <felix_fetche@hotmail.com>
Trimis: marți, februarie 19, 2019 11:28 p.m.
Către: martinbogo/i2-battery-diagnostics; martinbogo/i2-battery-diagnostics
Cc: Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Is not important the display thing, ...
Retriving correctly the datas from BCM and then filter and process them is the key of the job...

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: krex4 <notifications@github.com>
Trimis: marți, februarie 19, 2019 11:23 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Ok , thank you 👍😁

Enviado do meu iPhone

https://www.amazon.com/gp/product/B07D7ZQVDR/

felixfetche, 2019-02-19:
Only packs Certified by CRC are valid,
Only SN# and failure registers are “wellcomed” without CRC

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: felix Fetche <felix_fetche@hotmail.com>
Trimis: marți, februarie 19, 2019 11:29 p.m.
Către: martinbogo/i2-battery-diagnostics; martinbogo/i2-battery-diagnostics
Cc: Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

First byte, the CRC is almost most important, not easy, that’s why i do not let it for free

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: felix Fetche <felix_fetche@hotmail.com>
Trimis: marți, februarie 19, 2019 11:28 p.m.
Către: martinbogo/i2-battery-diagnostics; martinbogo/i2-battery-diagnostics
Cc: Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Is not important the display thing, ...
Retriving correctly the datas from BCM and then filter and process them is the key of the job...

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: krex4 <notifications@github.com>
Trimis: marți, februarie 19, 2019 11:23 p.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Ok , thank you 👍😁

Enviado do meu iPhone

https://www.amazon.com/gp/product/B07D7ZQVDR/

martinbogo, 2019-02-19:
Felix,

Last warning, I mean it.  If you're not going to contribute to the code
base, please refrain from commenting.  Bluntly "Be helpful, or go
elsewhere."

If you want to keep your closed-source solution private, that's fine.
You're more than allowed to do so.  However, this is an open source
project, and a collaborative one, meant to distribute the information
_freely_ per the MIT license.

Sincerely,
Martin

felixfetche, 2019-02-19:
Ok,
Man, you go fuck yourself elsewhere, you fill my mail with boolshits,
Unsubscribe me, sincerly not courious about your unprofesional job

Obțineți Outlook pentru iOS<https://aka.ms/o0ukef>

________________________________
De la: Martin Bogomolni <notifications@github.com>
Trimis: miercuri, februarie 20, 2019 12:39 a.m.
Către: martinbogo/i2-battery-diagnostics
Cc: felixfetche; Comment
Subiect: Re: [martinbogo/i2-battery-diagnostics] LCD on Arduino To read values (#3)

Felix,

Last warning, I mean it. If you're not going to contribute to the code
base, please refrain from commenting. Bluntly "Be helpful, or go
elsewhere."

If you want to keep your closed-source solution private, that's fine.
You're more than allowed to do so. However, this is an open source
project, and a collaborative one, meant to distribute the information
_freely_ per the MIT license.

Sincerely,
Martin

martinbogo, 2019-02-19:
@krex4 

I am now adding code to the project, that will simply display the values in a continuous loop.  You will need one of the displays I have linked to on Amazon, and you will need to connect SDA/SCL correctly both to the Segway battery and the display.

I'll be adding it into a "i2c display" branch, and once I've finished it will be merged into the main codebase with a #define to enable/disable the feature.

-Martin

rhinomods, 2019-02-20:
See I thought that was the problem that felix did not understand that he
subscribed to the githubs notifications, but didnt want to insult his
intelligence about not knowing how github works, we were not asking him
questions only replying to github.

krex4, 2019-02-20:
Ok , thank you very much 😁👍

Enviado do meu iPhone

martinbogo, 2019-02-21:
I believe there is a pattern to the "checksum" that involves adding the two bytes together then performing a checksum alg.

```
00111000 00010000 00110001 38 10 31
00111000 00010001 00110000 38 11 30
0x10+0x31 = 0x41
0x11+0x30 = 0x41
```

They have an identical "checksum" of 0x41 ( 65 ) 
This happens again here, but in a different way

```
10111010 00001110 00110001 BA E 31
10111010 00001111 00110000 BA F 30
0xE + 0x31 = 0x3F
0xF + 0x30 = 0x3F
```

Checksum is identical 0xBA ( 186 )

```
00111100 00000100 00111001 3C 4 39
00111100 00000110 00110111 3C 6 37
0x4 + 0x39 = 0x3D
0x6 + 0x37 = 0x3D
```
Checksum is identical 0x3C ( 60 )

Here are three with identical checksums:

```
10101010 00011001 00110110 AA 19 36
10101010 00011011 00110100 AA 1B 34
10101010 00000000 01001111 AA 0 4F

0x19 + 0x36 = 0x4F
0x1B + 0x34 = 0x4F
0x0 + 0x4F = 0x4F
```

Checksum is an identical 0xAA ( 170 )

SO -- this is definitely "A THING" -- the checksum is based on the addition of the first and second byte. 

Then we look at values that are off by one-- and things get a bit harder to predict:

```
01000001 00001000 00110000 41 8 30
01000000 00001001 00110000 40 9 30
0x8+0x30 = 0x38 = 56
0x9+0x30 = 0x39 = 57
```

They have "checksums" that are off by one 0x41, 0x40 but the checksum is one LESS for the higher number

```
00101101 00011100 00110000 2D 1C 30
00101100 00011101 00110000 2C 1D 30
0x1C + 0x30 = 0x4C = 76
0x1D + 0x30 = 0x4D = 77
```

Again, checksums are off by one ( 2D, 2C ) but this time the checksum is one HIGHER for the higher number

```
10000011 00100001 01010101 83 21 55
00000010 00100010 01010101 2 22 55
21+55 = 76
22+55 = 77
```

This has a "checksum" that is off by more than 1, so breaks the previous pattern. 0x83 (131) and 0x2 (2)

martinbogo, 2019-02-21:
Thanks to my friend Kristin, we have solved the checksum algorithm!

Take all bytes, add them together, then take modulo 8 and check if the value is 1.  If not, there may have been an issue in packet transmission.

closed 2019-11-08 by martinbogo