martinbogo/pt-battery-diagnostics #5: NiMH Batteries

moo1426, 2020-05-08:
Hi Martin,

I have 6 NiMH batteries I'm looking to pull data off. One of them still works normal, the others are all basically stone dead. Are you still looking for info from a working NiMH? If not do you know if the pinout is the same for communication? (aside from the +/- terminals)

Thanks!

MartinGrotendag, 2020-05-11:
Martin,

i only have 2 NiMH batteries but can't talk to them, (or at least one, have not tried the other yet.
Works fine for Lithium but no answer from NiMH, can't figure out why
I ran a I2C scan and to my surprise the NiMH seems to live at $31 as well, in config.h you write NiMH should live at $62 but as far as i can see it does not, anyway the BMS is not answering (will connect scope to see if any reply is visible) also noticed  TSENSORS 4 for NiMH batteries but the have 6 NTC's .. i thought i read somewhere that you could test on both types ..? did you ever do a successful read on NiMH batteries ..?
thanks in advance for your answer,

Martin

nazar404, 2020-05-11:
Martin , 

The BMS are mostly bad .  I do know someone with batteries for sale at least last month .
I think he around 6 packs . 

Hopefully this helps 

Neil

martinbogo, 2020-05-11:
Hello Martin Grotendag!

( Your parents had excellent taste in first names, by the way. )

I haven't done extensive testing with the NiMH batteries as I only have
Segway a i2 and i2se which only use the Lithium cells.   Interesting that
the NiMH battery lives at $31!  As you discover more please feel free to
modify the code and I'll merge in your changes.

Sincerely,
Martin B.

martinbogo, 2020-05-11:
When NiMH and LiION batteries stop responding entirely ( even when you
apply 12V to the battery BMS ) then those are effectively completely dead.
I would take those to the recycler -- please avoid putting them in the
refuse if at all possible.

I have a LiION battery that accidentally shorted across the 72V terminals,
and while the BMS is alive the battery needs to have the fuse replaced.
The terrible thing is that there's no simple way to open the batteries just
to replace the fuse which is for me a terible design flaw.  There is a
company in Shenzhen that makes batteries with two O-ring seals and steel
screws and heat-set inserts -- but I haven't seen them on AliBaba for a
while and I'm afraid the Segway secondary battery market has pretty much
dried up.

Unfortunately Segway/Ninebot is a company that is in decline now that the
demand for scooters has dropped out of the market. ( Bird/Lyft and so many
other scooter companies are exiting )

moo1426, 2020-05-11:
> Martin , The BMS are mostly bad . I do know someone with batteries for sale at least last month . I think he around 6 packs . Hopefully this helps Neil
> […](#)
> Sent from my iPhone

Neil, If they still have the batteries I might be interested, it looks like 3/4 of my BMS's are dead so I'd be in for a set or 2 at the right price

nazar404, 2020-05-11:
Sorry about the typos this morning as auto correct kicked in on my iphone , if you need NiMH batteries I can help you find some .

I think if you modify the address code and apply 12 volts to the BMS to wake up the BMS.

good luck 

Neil

nazar404, 2020-05-11:
> … <x-msg://9/#>

Kory Bucklew 
317-371-0265 

give him a call

martinbogo, 2020-05-11:
Hey folks,

The "issues" threads are for software bugs and development.  If you want to
talk about buying/selling/trading batteries and components, please email
the person directly.

Sincerely,
Martin ( The Author/Coder )

MartinGrotendag, 2020-05-11:
Neil,

a empty mail just escaped ..;-) ,,ignore it .. , tnx for your answer, yes buying new batteries is of course te way with the least resistance, just put two Lithium packs under the 167 and of course it runs fine ..bought and mounted all new NiMH cells and the unit did not fire up that’s why i downloaded (your..?) sketch … have an extra device in the I2C bus (a display lives at $27 but the weird thing is that i do (so now and then) find the BMS but not on 62 as should be according the sketch but at 31 (the address for Lithium.
As written i CAN talk with a Lithium cell at address 31 but CAN’T with the NiMH cell ..God knows why, have to look what BMS answers with a login analyses .. bit complex to figure out on a scope ..;-)

Did you or anyone ever connected a Arduino (or whatever) successfully to a LiMH battery pack ..? and if so at what address ..?

Just out of curiosity what batteries NiMH/Lithium ? in what state ..? price ..? and where ..? (transport..?)

Martin

MartinGrotendag, 2020-05-11:
Martin,

tnx for answering,

I prefer to bring it back to fife so a working BMS (even if i have to open this horrible case .;-) would be nice as written renewed all cells but one of the BMS es answers at air 31 (not the 62 in the sketch) and the other refuses to talk to me at all, has for sure not been mistreated so no idea why ..

So if i can buy a LiHM BMS that works (really wonder at what address it answers;-) yes that would help to have at least two BMSses that the Segway can talk to ..

Martin

MartinGrotendag, 2020-05-11:
Martin,

As you can see Lithium works fine at 31 NiMH reports for duty but does not answer at all, the other does not even answer ..

Martin

[cid:F4DBC6D9-3201-4518-A11B-8B6004DAEBEF@fritz.box]

[cid:49CBCF24-FE3B-4C52-B9E8-9AD66609476F@fritz.box]

MartinGrotendag, 2020-05-13:
Martin,

the does the “blown fuse” NiMH battery communicate with our software ? if not could you please peak to see if you can see why, should be easier for you since you wrote the software and would help me a lot.

As written i have two LiMH batteries with new cells but Segway refuses to fire up, this would be logical if one of the batteries refuses to communicate, the reason i would like to talk with them is because the guy that put the new cells in (friend of me) broke two of the six NTC’s and decided to put new ones in both batteries, tried to figure out what NTC’s there are used in the batteries and bough 15 new ones, when i measure the old vs the new ones these is a small difference resulting in 1,58 Volt over the new ones at room temperatuur and 1,23 Volt over the original ones.

This alone is 28% higher and i have not even looked if they’re hanging in a Opamp circuit to come in better range for the (5 Volt is guess) AD converter, if so the error wil be amplified as well resulting in a BMS telling the Segway that the batteries are 48 degrees Celsius, enough reason for the Segway not to fire up ..;-) so knowing what the BMS is “thinking” is essential here…

The I167 runs fine with (borrowed) Lithium cells but i would very much like to have the NiMH batteries working again especially because they have all been fitted with new celles ...

If i measure a number of old NTC’s the have quite some spread as well so i fear that the NTC’s are calibrated in software in the BMS .. if so replacing them all is the most stupid thing he could have done, i will never (other than trail and error) find back what NTC should be where ..

I heve not looked in depth to the BMS circuit board they’re covert in a kind if thick vernis so water resistant and hard to see and measure ..

As you can see, i would be very happy if i could look in the head of the BMS telling me what voltages it measures and what temperature …

Finally, if i appear to have one BMS that is willing to talk to me and one that refuses to do so i would be very happy if i could buy you broken fuse NiMH battery because, as you know the BMSses in the Lithium batteries are totally different.

so please let me know if you’re able to talk to your battery (i assume you don’t have to open it but can communicate with the BMS simply by powering the BMS with 12 volts as you do for the Lithiums as well..

If you appear to be able to communicate with it i’m really curious at what address (i used a I2C scanner from Github) it answers, mine does at $31 not $62 (one bit difference by the way..;-) so yes at least the correct address would be nice and even better a way to read the data form the, at least answering to address poll one ..

and if the not answering one is beyond repair (or even when not) i would be interested in buying your one for the BMS system, assuming t “talks” ..;-)

thanks in advance for your help, and stay safe,

Martin

closed 2020-11-20 by martinbogo