martinbogo/pt-battery-diagnostics #6: Hi!

Migu5499, 2020-06-28:
Hi idk how thia project works, i just want to build a segway battery diagnostic that tell me if cells are good enough, there is a list of materials and processes i have to do to build it??.
Thank for your work, hope we can talk!

martinbogo, 2020-06-28:
Migu...

This is code that I wrote to document and help people understand the Segway
I2 lithium battery BMS responses.

It's useful, but not meant to be a product... just a guide and a learning
tool.

To use the tool you're going to need an Arduino, and some electronics
engineering skis and confidence working around high voltage (70v)
batteries.  You can seriously hurt yourself and damage your equipment.

Migu5499, 2020-06-28:
Nice thank you for your fast reply, im a e-scooter fixer so i have knowledge about high voltage, but im a little noob building this things but sure i want to try build a segway battery checker!!

closed 2020-11-20 by martinbogo