martinbogo/pt-battery-diagnostics #8: Reset BMS?

kitekara, 2020-12-01:
Hi! Read trough this project, could not tell: have you found a way resetting the BMS when locked down?

martinbogo, 2020-12-01:
Define "locked down"?

kitekara, 2020-12-01:
Red light on segway even if cells are balanced to 3.00v each. Have done some research telling me that BMS would need restart/recalibrating if so.

kitekara, 2020-12-01:
Will not take charge. Will not power on.

nazar404, 2020-12-01:
Been having good luck , once the batteries are past 55volts

martinbogo, 2020-12-01:
Red light generally means your battery has fallen below safe voltage... or
you are experiencing the Rev AF/AG bug where the board locks up for no
reason.

kitekara, 2020-12-01:
3x 23= 69V

martinbogo, 2020-12-01:
Once you get a battery past 50V, the BMS should recover and light up.
There's no "lock" that I'm aware of but if it fails a safety check ( cell
voltage low even if pack is at 50v ) then it will stay in "red light mode"
 Sorry I can't be of more help.

kitekara, 2020-12-01:
Ok, thanks anyway!

martinbogo, 2020-12-01:
My only other suggestion -- get the battery pack up to ~55V then connect it
to a segway I2/X2 or a Segway off-board charging unit.  I didn't bother to
try decoding more than the battery diagnostics -- but there is a command
stream that the Segway charge controller sends to a battery when it's first
plugged in.  That might reset the battery error state.

kitekara, 2020-12-01:
Okay...,hmm. My battery are currently about 70V, have connected to an I2 several times, still red light. Playing with the plug i got green light for one second or two, guess some of the connector did not connect. Have tried 12v on j5. Have tried 73,6v j2/j4 for some shorter time. Am thinking about disconnect battery cell cables and go with a power supply directly to board..

kitekara, 2020-12-01:
There is a service option you can from segway fixing companies. They offer to reset/ recalibrate your BMS for about 200dollar

martinbogo, 2020-12-01:
You mean like the "segway reviver" that ... all it did ... was provide a
constant current power supply source at 50-100mA to gently bring a battery
back to 50V?

There's a lot of bullshit and snake-oil out there... that was the point of
reverse engineering the protocol and being able to communicate with the BMS
in the first place.

I may not have yet decoded every last little piece of the puzzle, but I got
most of it!   I don't think those companies do very much, frankly.
Certainly not $200 worth of work.

nazar404, 2020-12-01:
Probably a few bad cells inside the battery , you can use Martin software to check  or the bms has failed .

Foxtrot81355, 2024-06-25:
> Хорошо ..., хм. В моей батарее в настоящее время около 70 В, несколько раз подключался к I2, все еще горит красный индикатор. Играя со штекером, я на секунду или две загорелся зеленый свет, думаю, какой-то разъем не подключался. Пробовал 12 В на j5. Пробовал 73,6 В на j2 / j4 в течение более короткого времени. Я подумываю о том, чтобы отсоединить кабели аккумуляторных элементов и подключить источник питания непосредственно к плате..

Hello. I have the same problem, that's why I'm here. Have you found a solution? We tried to discharge one battery to its nominal value, but the indicator was still red and blinked green once at first. At the same time, the battery is balanced.

Foxtrot81355, 2024-06-26:
> Красный индикатор обычно означает, что напряжение вашей батареи упало ниже безопасного уровня ... или вы столкнулись с ошибкой Rev AF / AG, когда плата блокируется без причины.
> […](#)

Can I tell you more about this? I changed all the 18650 elements, equalized the voltage, but there is still an error and red lights. Is there any solution to get out of this situation?