martinbogo/pt-battery-diagnostics #9: Completely dead battery checking

urx, 2021-01-29:
Hello.
I have a Segway XT with completelly dead batteries - 0V, cracked cells and so on. I am in a half way of ordering new cells for it, but I know nothing about BMS state, may be BMS is broken too. With this project can I communicate with BMS without battery pack? Can I use regulated power source to turn BMS on somehow?

PS: This is very interesting project! Can I participate somehow? I have 2 dead batteries, oscilloscope, a lot of microcontrollers (stm32, stm8, atmega, raspbery and so on), logic analyzer and soldering station. May be I can be of use :)

PPS: sorry for my poor english )

martinbogo, 2021-01-29:
I'll be honest -- if you have a battery with 0V that has been that way for
a long time -- it's VERY LIKELY that the batteries have leaked, and that
your BMS will have suffered damage/corrosion from the electrolyte.   This
is the case with almost every single 0V / cracked cell battery I have
analyzed.  I've done about 50 over time, so it's pretty statistically
certain.

The code I've provided is a tool to help you determine the status of a
pack... and let you dump interesting information about cells to determine
which cell group might be bad in a mostly-ok pack, or the overall state of
a pack.

To quote Miracle Max from The Princess Bride:"With all dead, well, with all
dead there's usually only one thing you can do." ... "Go through the
pockets and look for loose change."

Although it's painful because the batteries aren't cheap -- you should
refill this pack ONLY if the BMS looks absolutely clean.  If you give the
BMS 12V to the activation pin, and use the i2c bus to read it, you should
still be able to read diagnostics... however there is a state the BMS can
enter which disables itself ( red light blink ) which you may not be able
to recover from.

urx, 2021-01-29:
Unfortunately I don't know BMS state (red blink or not) because I am unable to turn XT on with completely dead both batteries.

I have battery disassembled and battery pack removed from it. PCB itself looks good, no signs of corrosion.