martinbogo/pt-battery-diagnostics #10: Reviving a battery

tit0n, 2021-03-05:
Hi, first, thanks for your contribution and making this project open source. There are so many snakeoil and way over priced tools for the Segway community...

It might be a silly question, but do you have any details on how to revive a battery than hasn’t been charged in a few months. It’s is currently at around 8v and solid red when trying to charge it with onboard charger. 

I understood that the batterie need to somehow go back to ~50v to be charged with the onboard charger. I have seen a few companies offering to revive the battery for $300 or more. But it seems to be way over priced. Would you happen to know what those reviver do exactly ? And could I build a cheap DIY reviver myself without damaging my battery ?

bottleworks, 2021-03-05:
You can’t revive an 8 volt battery. The cells are damaged by discharging it that low.

tit0n, 2021-03-05:
> You can’t revive an 8 volt battery. The cells are damaged by discharging it that low.

eh. Not really the answer I was hoping to read. :)

The battery was rebuilt ~6 months ago. I must have charged it 2-3 times and it worked fine. 
I didn't charge it for the past 3 months... Both batteries were left in the PT (gen1). 
One battery is still at 75v, while the other is at 8v now. Any idea what could have happened ?

martinbogo, 2021-03-05:
8V is too low... you definitely have damaged cells at this point.

While I very much DO NOT RECOMMEND THIS... You can try trickle charging
with constant current ( 50mA-100mA ) the battery to around 50-51V cutoff.

That will bring the BMS back to 'safe voltage' mode and then you can
connect up to I2C and power it up to evaluate the cell groups.  If they
appear to be charged and stable... you can try charging with an off board
charger.

Again. *** I REALLY REALLY DON'T RECOMMEND THIS ***

closed 2022-06-09 by martinbogo