martinbogo/pt-battery-diagnostics #11: Code Arduino Diagnostic Battery Segway

krafaello, 2021-03-24:
Hi
I  Would like chceck my battery Segway -secection battery.
I have problem. I write code my Arduino uno but program dont work.
I have LCD 2004 but nothing is displayed. The code is 100% correct?

martinbogo, 2021-03-25:
Last time I checked, the code compiled and worked.  You can communicate
using the serial menu if you're having issues with the LCD code.

krafaello, 2021-03-25:
Hi
My port Lcd 2004 is 0x27

martinbogo, 2021-03-25:
Sounds like you can fix the code then, easily.  Good luck!

http://wiki.sunfounder.cc/index.php?title=I2C_LCD2004

krafaello, 2021-03-27:
Hi 
My LCD is good but (positive test).
When write Code Diagnostic Battery  received „Bushes”.###########.
Serial menu shows good...
What I doping wrong?
You can check again code?
Thanks 
[image]
[image]
Raf

martinbogo, 2021-03-27:
I am no longer actively developing this code... I'm afraid I don't have the
time to debug your issue.  Sorry.

closed 2022-06-09 by martinbogo