martinbogo/pt-battery-diagnostics #12: Pt Bms emulator?

EngFix, 2021-06-25:
Man i am a really huge fan of your work, and as i can see you helped a lot of pt's battery around the globe.
But i really must ask, would be possible to emulate the pt bms with a arduino??
So it would be possible to use a generic bms and an emulator?

martinbogo, 2021-06-25:
Honestly, I don't know.  That's very much beyond the scope of anything I
ever researched.  There are Chinese BMS makers, so I know someone
eventually either ripped off the code or reverse engineered it.

EngFix, 2021-06-27:
First huge thank you for answering,

There is chinese BMS makers like yybms but prices are absurd.
(i´m from Brazil so add 100% taxes 😩, shipping and 3world country bull  shit)
Not to bother you but if ever cross your mind something like this.....
My ideia was use a emulator to build a battery out of old laptop cells, regular bms and emulate the original BMS .
something like the ninebot.
https://forum.electricunicycle.org/topic/17858-ninebot-z-series-bms-emulatorconverter/

I have little knowledge on programing but huge on eletronics, battery building ..........and looking for a solution.
in any case , im really thankful for your work and answer.
[https://content.invisioncic.com/r254450/monthly_2015_05/meta_image.png.60d098e5dcbd57022bda5262ed6bebb5.png]<https://forum.electricunicycle.org/topic/17858-ninebot-z-series-bms-emulatorconverter/>
Ninebot Z series BMS emulator/converter - Ninebot - Electric Unicycle Forum (euc community)<https://forum.electricunicycle.org/topic/17858-ninebot-z-series-bms-emulatorconverter/>
Hi all, Many users of Ninebot Z10 knows that if our BMS will be burnt - that is a big problem, and you need to buy new battery pack. I built some protocol emulator/converter which allows to use some third-party BMS like this. It is well known series of BMS, which prove their durability and flexibility.
forum.electricunicycle.org

closed 2022-06-09 by martinbogo