martinbogo/pt-battery-diagnostics #14: a 3D printed battery connector

long16652, 2023-04-23:
If you need a 3D printed battery connector made, I highly recommend the one Rotek designed here on Shapeways. Well worth supporting his work! https://www.shapeways.com/product/EN759FWRC/segway-battery-conector

Hello, after reading your source code, I am very interested and have also purchased an Arduino device. However, I cannot find the 3D file for this 3D printed battery connector. Could you please provide one? Thank you. My email address is: 35890985@qq.com

toddsanders, 2023-05-25:
It is the battery_connector.stl file hosted on this GitHub, I plan to print it today.

long16652, 2023-05-25:
battery connetor.stl我已经3D打印过了几个，孔位有点不准确，而且里面的金属插针不太好匹配，所以我自己建模了一个，并且打印了，设定了金属插针的尺寸。

closed 2023-09-04 by martinbogo