martinbogo/pt-battery-diagnostics #15: photo_of_my_messy_setup.png evtl. shows minus Cable on wrong port

secman248, 2024-06-24:
Hi Martin,
i discoverd your code  an i am very happy to give it a try. Thank you for your work an for making it public. I already ordererd all the necessary hardware an am waiting for it to arrive. In the meantime i try to find out what cables need to go where based on the infos you provided. I am a bit confused, as you wrote Segway battery J4 should go to Arduino GND. But i am almost shure that on the Photo its going to the 5V connector of the Arduino. Can you plaes explain. Thanks, Georg

martinbogo, 2024-06-24:
It's been _**years**_ since I last touched this code or looked at the diagrams.  No matter what, you should be VERY CAREFUL about your setup.  The live POS(+) terminal has 74V and will happily fry anything you connect to it as solidly as any welder can -- while at the same time popping a fuse inside the battery that simply cannot be reached without completely disassembling the battery -- _**consider yourself warned**_

I think the photo I took was using one of my dead Segway batteries.  It's entirely possible I had it plugged into the wrong pin on that arduino ( which shouldn't do much to an arduino -- 5V arduino to ground on the battery will do very little -- not a lot of current to be had there )

But you're going to want to make sure it's connected to a GND pin.  You're going to need a common ground to make sure your i2c signals come in clear and strongly with a common ground for the floor of the signal.

closed 2024-06-24 by martinbogo

secman248, 2024-06-24:
Thanky you very much. You are absolutely right by saying "you should be VERY CAREFUL about your setup", thats exactly why i asked. So i am going with ground. I am very exited with your code., it will probably help me a lot. Thank you !

nazar404, 2024-06-24:
His code works great ..  it has been 4 yrs since I used it ..  Sent from my iPhone

martinbogo, 2024-06-24:
Thanks Nazar...

I wish I had a reason to continue developing it.  What I put together
though is at least a good base for other people who want to explore the
battery BMS and what the communication protocol is.

There are some parts I never finished decoding -- charge/discharge rates,
serial number, production date.. that kind of thing.  It's in there, and
another person ( who made a closed source solution ) named Felix managed to
figure it out.  It's just a question of some patient analysis, time, and
hacking.

martinbogo, 2024-06-24:
Side note!  I don't think I implemented this fully:

register 29, contains soc means "state of charge"
register 212 contains amp means charge/discharge current

soc=(int)((k_avg_soc*raw_avg_soc[2])/254);
if(summ_volt<76.5 && raw_avg_soc[2]==254) soc=0; //Case of Under Level SOC

amp=(256*(raw_avg_amp[1])+raw_avg_amp[2])/k_cell_amp;

float k_avg_soc=100;
float k_cell_amp=128;

secman248, 2024-06-24:
cool. i try my best to understand and implement this stuff.

nazar404, 2024-06-25:
I have Felix product as well ,  it worked well as well !!  But I no time to update Segway batteries anymore ,  as I have around 40 plus core to update with new cells ..  maybe one day Sent from my iPhone