martinbogo/pt-battery-diagnostics #16: Reading the battery board without cells

Foxtrot81355, 2024-06-25:
Hello. I have a couple of questions. But first, I would like to express my gratitude.Your project is very interesting! This is a colossal job! You did a great job!
When trying to read the battery, which is currently low, there is no response from the board. Is this how it should be if there are no connected cells? No information can be obtained from the battery. At the same time, the battery "Woke up" and the light bulb on the board glows.
What is the function of the display? Is he the only one connecting?
[image]
[image]

martinbogo, 2024-06-25:
I didn't quite finish the display code.  At the time, it was just to
attempt to make it a little easier to see the results without a laptop
nearby.

For my own use, I only ever really configured SERIAL_SHELL in config.h

Foxtrot81355, 2024-06-26:
[image]
I uploaded your code to the arduino uno connected according to this scheme, the batteries did not connect to 4 boards, what could I have done wrong?

martinbogo, 2024-06-26:
Looks correct to me.  I don't tend to use Vin to feed 12V to the Segway
battery to wake it up though... The arduino can source very little current
on that pin ( 30mA, MAX ) so you may want to try using an independent 12V
battery pack the way I do ( or a 12V lead-acid gel cell, like is used in
UPS or alarm systems? )

Foxtrot81355, 2024-06-27:
> На мой взгляд, выглядит правильно. Однако я не склонен использовать Vin для подачи напряжения 12 В на аккумулятор Segway, чтобы включить его... arduino может подавать очень небольшой ток на этот вывод (не более 30 мА), поэтому вы можете попробовать использовать независимый аккумулятор на 12 В, как это делаю я (или свинцово-кислотный гелевый элемент на 12 В, который используется в ИБП или системах сигнализации?)
> […](#)

OK, thanks for the advice, it was originally connected separately. Why can't I connect the battery? When turned on, the battery board lights up red and green, but there is no response from the BMS board.. in the port stream, with any request from Arduino except help, it issues I2C Err 2 adr :С 
I bought a lot of 18650, and old batteries from Segway for repair, but nothing works..

martinbogo, 2024-06-27:
Error 2 generally means the chips is either not connected (or not
correctly) or the wrong address was sent.  I. no longer have any Segway
batteries here, so I honestly can't help much more than this, especially
not knowing the precise condition of your batteries or your setup.

I always started with a known-good battery to test my setup before moving
on to diagnose others.   Also, the last generation battery I had access to
was an "AH" and I know things may have changed over time.

secman248, 2024-06-28:
what  value do you have for Wire.setClock  in your SegwayBatteryDiag.ino ?
ist it Wire.setClock(400000);  or Wire.setClock(100000); ?
if it is 400000 try 100000

Foxtrot81355, 2024-07-11:
> какое значение у вас есть для Wire.setClock в вашем SegwayBatteryDiag.ino? это Wire.setClock (400000); или Wire.setClock (100000); ? если это 400000, попробуйте 100000

We tried different options, unfortunately there is no result. Most likely, the board is defective. There is no serviceable board to check.

justbob43, 2025-09-03:
I'm also trying to rebuild my batteries. I have Martin's code loaded into a MEGA2560 and can communicate with both a fully assembled, (dead), battery or an BMS board separately. I have ordered new cells and tried to install them to no avail.

bottleworks, 2025-09-03:
When you attach the cells to the BMS, you must start from 0V and work your way up to pack's positive.  If you attach the packs negative and positive leads first and then attach the balance leads after, you will damage the BMS.  Also, the two sub groups of cells can EASILY get flipped around and you'll burn out the board.

thaihyquanquy, 2025-09-07:
> I'm also trying to rebuild my batteries. I have Martin's code loaded into a MEGA2560 and can communicate with both a fully assembled, (dead), battery or an BMS board separately. I have ordered new cells and tried to install them to no avail.

It is so thankful that someone like Martin, you and any other fellows who sucessfully connect and diagose the batteries. Would you please share the setup that you are using with Mega2560? I have it hooked up with 3.5" TFT touch screen but don't know how to configure the code to make them work. Thanks indeed.